muvox-io/euphonium #1: get AI Thinker AudioKIT with ES8388 working

HeikoGr, 2022-01-01:
i did not manage to get the AudioKIT board working.

i tried to alter the i2s adress to 0x20 (as on my cspot driver) and adjust the gpios.
i wa not able to make ES8388 an option in the dropdown selector, so i even manipulate the AC101 driver (file) to match my ES8388 driver file content. without success. I don't know what else i could do with my limited capabilities.

feelfreelinux, 2022-01-01:
Hi. The I2S driver shifts the address by << 1, therefore in the berry part of the driver, the address still should be 0x10 `(0x10 << 1 == 0x20)`. You most likely only need to change the pin definitions. Feel free to reach out to me on any communicator, I would love to debug this further :)

HeikoGr, 2022-01-01:
ok, i changed my files again back to 0x10. no i need to wait till github is ready...

 (i managed to make your code working in idf 4.3 - i can't install idf 4.4 on wsl2 under windows. But no the web part didn't compile ... so i'm stuck with the github actions...)

i sent you a friend request on discord

HeikoGr, 2022-01-02:
it works

closed 2022-01-02 by HeikoGr