muvox-io/euphonium #9: Build Errors

shark8594, 2022-01-06:
Hi,

There seem to be a few build errors on the main branch, log below:

C:\euphonium\euphonium\targets\esp32\app>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\abhin\esp\custom_examples\euphonium\euphonium\targets\esp32\app\build
Executing "ninja all"...
[1/141] Building CXX object esp-idf/cspot/bindir/bell/CMakeFiles/bell.dir/src/platform/esp/MDNSService.cpp.obj
FAILED: esp-idf/cspot/bindir/bell/CMakeFiles/bell.dir/src/platform/esp/MDNSService.cpp.obj
ccache C:\Users\abhin\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DBELL_USE_MBEDTLS -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPB_ENABLE_MALLOC -DUSE_DEFAULT_STDLIB=1 -Iconfig -I../../../../euphonium/bell/nanopb -I../../../../euphonium/bell/include -I../../../../euphonium/bell/libhelix-aac -I../../../../euphonium/bell/libhelix-mp3 -I../../../../euphonium/bell/cJSON -I../../../../euphonium/bell/tremor -Iesp-idf/cspot/bindir/bell -I../../../../../../../components/newlib/platform_include -I../../../../../../../components/freertos/include -I../../../../../../../components/freertos/port/xtensa/include -I../../../../../../../components/esp_hw_support/include -I../../../../../../../components/esp_hw_support/include/soc -I../../../../../../../components/esp_hw_support/port/esp32/. -I../../../../../../../components/heap/include -I../../../../../../../components/log/include -I../../../../../../../components/lwip/include/apps -I../../../../../../../components/lwip/include/apps/sntp -I../../../../../../../components/lwip/lwip/src/include -I../../../../../../../components/lwip/port/esp32/include -I../../../../../../../components/lwip/port/esp32/include/arch -I../../../../../../../components/soc/include -I../../../../../../../components/soc/esp32/. -I../../../../../../../components/soc/esp32/include -I../../../../../../../components/hal/esp32/include -I../../../../../../../components/hal/include -I../../../../../../../components/esp_rom/include -I../../../../../../../components/esp_rom/esp32 -I../../../../../../../components/esp_common/include -I../../../../../../../components/esp_system/include -I../../../../../../../components/esp_system/port/soc -I../../../../../../../components/esp_system/port/public_compat -I../../../../../../../components/esp32/include -I../../../../../../../components/driver/include -I../../../../../../../components/driver/esp32/include -I../../../../../../../components/esp_pm/include -I../../../../../../../components/esp_ringbuf/include -I../../../../../../../components/efuse/include -I../../../../../../../components/efuse/esp32/include -I../../../../../../../components/xtensa/include -I../../../../../../../components/xtensa/esp32/include -I../../../../../../../components/vfs/include -I../../../../../../../components/esp_wifi/include -I../../../../../../../components/esp_wifi/esp32/include -I../../../../../../../components/esp_event/include -I../../../../../../../components/esp_netif/include -I../../../../../../../components/tcpip_adapter/include -I../../../../../../../components/app_trace/include -I../../../../../../../components/esp_timer/include -I../../../../../../../components/mbedtls/port/include -I../../../../../../../components/mbedtls/mbedtls/include -I../../../../../../../components/mbedtls/esp_crt_bundle/include -I../../../../../../../components/pthread/include -I../../../../../../../components/mdns/include -I../../../../../../../components/console -mlongcalls -Wno-frame-address -g   -Wunused-const-variable -Wchar-subscripts -Wunused-label -Wmaybe-uninitialized -Wmisleading-indentation -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Os -freorder-blocks -fmacro-prefix-map=C:/Users/abhin/esp/Custom_examples/euphonium/euphonium/targets/esp32/app=. -fmacro-prefix-map=C:/Users/abhin/esp=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fexceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-1404-gc13afea63-dirty\" -DESP_PLATFORM -DBELL_USE_MBEDTLS -std=gnu++17 -MD -MT esp-idf/cspot/bindir/bell/CMakeFiles/bell.dir/src/platform/esp/MDNSService.cpp.obj -MF esp-idf\cspot\bindir\bell\CMakeFiles\bell.dir\src\platform\esp\MDNSService.cpp.obj.d -o esp-idf/cspot/bindir/bell/CMakeFiles/bell.dir/src/platform/esp/MDNSService.cpp.obj -c C:/Users/abhin/esp/Custom_examples/euphonium/euphonium/euphonium/bell/src/platform/esp/MDNSService.cpp
C:/Users/abhin/esp/Custom_examples/euphonium/euphonium/euphonium/bell/src/platform/esp/MDNSService.cpp:1:10: fatal error: MDNSService.h: No such file or directory
 #include "MDNSService.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
[3/141] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\abhin\esp\Custom_examples\euphonium\euphonium\targets\esp32\app\build\bootloader\esp-idf\esptool_py && C:\Users\abhin\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe C:/Users/abhin/esp/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/abhin/esp/Custom_examples/euphonium/euphonium/targets/esp32/app/build/bootloader/bootloader.bin"
Bootloader binary size 0x69c0 bytes. 0x640 bytes (6%) free.
[12/139] Building CXX object esp-idf/cspot/bindir/bell/CMakeFiles/bell.dir/src/platform/esp/TLSSocket.cpp.obj
In file included from ../../../../euphonium/bell/include/platform/TLSSocket.h:4,
                 from C:/Users/abhin/esp/Custom_examples/euphonium/euphonium/euphonium/bell/src/platform/esp/TLSSocket.cpp:1:
../../../../euphonium/bell/include/BellLogger.h: In member function 'void bell::BellLogger::printFilename(std::__cxx11::string)':
../../../../euphonium/bell/include/BellLogger.h:86:17: warning: unused variable 'c' [-Wunused-variable]
             int c;
                 ^
[13/139] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/driver/I2SDriver.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/driver/I2SDriver.cpp.obj
ccache C:\Users\abhin\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DEUPH_VERSION=\"debug\" -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../main/driver -I../main/bluetooth -I../main/ota -I../../../../../../../components/newlib/platform_include -I../../../../../../../components/freertos/include -I../../../../../../../components/freertos/port/xtensa/include -I../../../../../../../components/esp_hw_support/include -I../../../../../../../components/esp_hw_support/include/soc -I../../../../../../../components/esp_hw_support/port/esp32/. -I../../../../../../../components/heap/include -I../../../../../../../components/log/include -I../../../../../../../components/lwip/include/apps -I../../../../../../../components/lwip/include/apps/sntp -I../../../../../../../components/lwip/lwip/src/include -I../../../../../../../components/lwip/port/esp32/include -I../../../../../../../components/lwip/port/esp32/include/arch -I../../../../../../../components/soc/include -I../../../../../../../components/soc/esp32/. -I../../../../../../../components/soc/esp32/include -I../../../../../../../components/hal/esp32/include -I../../../../../../../components/hal/include -I../../../../../../../components/esp_rom/include -I../../../../../../../components/esp_rom/esp32 -I../../../../../../../components/esp_common/include -I../../../../../../../components/esp_system/include -I../../../../../../../components/esp_system/port/soc -I../../../../../../../components/esp_system/port/public_compat -I../../../../../../../components/esp32/include -I../../../../../../../components/driver/include -I../../../../../../../components/driver/esp32/include -I../../../../../../../components/esp_pm/include -I../../../../../../../components/esp_ringbuf/include -I../../../../../../../components/efuse/include -I../../../../../../../components/efuse/esp32/include -I../../../../../../../components/xtensa/include -I../../../../../../../components/xtensa/esp32/include -I../../../../../../../components/vfs/include -I../../../../../../../components/esp_wifi/include -I../../../../../../../components/esp_wifi/esp32/include -I../../../../../../../components/esp_event/include -I../../../../../../../components/esp_netif/include -I../../../../../../../components/tcp [... cut ...]
../../../../euphonium/bell/include/CryptoMbedTLS.h: At global scope:
../../../../euphonium/bell/include/CryptoMbedTLS.h:35:22: warning: 'DHGenerator' defined but not used [-Wunused-variable]
 static unsigned char DHGenerator[1] = {2};
                      ^~~~~~~~~~~
../../../../euphonium/bell/include/CryptoMbedTLS.h:20:22: warning: 'DHPrime' defined but not used [-Wunused-variable]
 static unsigned char DHPrime[] = {
                      ^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

HeikoGr, 2022-01-06:
Can you try the following(?):
- replace the bell submodule folder withe the most recent one
- replace 'vTaskYield' with 'taskYIELD' 

i think it was this changes i got it compiling. But i wasn't able to make a pull request yesterday.

shark8594, 2022-01-06:
Thanks build works now :). Additional changes:
1. Create spiffs folder in targets/esp32/app/
2. Remove the type casting in I2SDriver.cpp:28:24

closed 2022-01-06 by shark8594