muvox-io/euphonium #14: Littlefs not building

shark8594, 2022-01-11:
Hi,

The managed components do not seem like they are part of the Cmakelists, causing the cmake to fail:

:/Users/abhin/esp/components/wear_levelling C:/Users/abhin/esp/components/wifi_provisioning C:/Users/abhin/esp/components/wpa_supplicant C:/Users/abhin/esp/components/xtensa
CMake Error at CMakeLists.txt:10 (littlefs_create_partition_image):
  Unknown CMake command "littlefs_create_partition_image".

-- Configuring incomplete, errors occurred!

Thanks!

HeikoGr, 2022-01-12:
I managed to build euphonium with the use of WSL2 this way:
https://github.com/HeikoGr/euphonium/wiki/using-Windows-Subsystem-for-Linux-(WSL)-for-compilation

it‘s way easier then setup a Windows only toolchain…

shark8594, 2022-01-13:
Thanks for the steps, works in ubuntu.

closed 2022-01-13 by shark8594