muvox-io/euphonium #19: Buffering changes

shark8594, 2022-01-21:
Hi,

Have been testing out euphonium with a custom audio board (ES8388) and it works great!. In the previous spotify stand alone version cspot, I could tweak the buffer size and logic in ChunkedAudioStream (BUFFER_SIZE), however in the current project seems like it is moved to ChunkedByteStream? Any clues on how to change the buffer size and add logic? Also, I observed a segfault when we skip/seek tracks on spotify. 

Thanks!

feelfreelinux, 2022-01-22:
Cspot crashes are currently being worked on, current master is not really stable in that regard :). Euphonium has it's buffer defined in `MainAudioBuffer.h`

shark8594, 2022-01-22:
Thanks :), I tried it and seems like the change is taking effect:
[image]

closed 2022-01-22 by shark8594