muvox-io/euphonium #21: Android app build fail

shark8594, 2022-02-15:
Hi,

Seems like the app is missing some sub-modules and the build is failing.  Error below:

FAILURE: Build failed with an exception.

* Where:
Script 'C:\Workspace\euphonium_tip\euphonium\app\android\app\capacitor.build.gradle' line: 10

* What went wrong:
A problem occurred evaluating script.
> Could not read script 'C:\Workspace\euphonium_tip\euphonium\app\android\capacitor-cordova-android-plugins\cordova.variables.gradle' as it does not exist.

Help greatly appreciated :)

Thanks!

shark8594, 2022-02-15:
Adding this resolved it:

npx cap add android 

Thanks!

closed 2022-02-15 by shark8594

danielehrhardt, 2022-02-15:
You also can download the apk here:
https://github.com/feelfreelinux/euphonium/actions/runs/1836786181

shark8594, 2022-02-16:
Thanks, great work!