muvox-io/euphonium #23: Getting started issues with ai esp32 audio kit 2.2

rhaamo, 2022-03-08:
Hello,

Just got an Ai ESP32 Audio Kit V2.2 and flashed it with latest tarball from the releases but I have some issue.
- Nothing out of the earphones TRS jack
- Line in from TRS jack is not handled ?

No matter what board I use in the selector, nothing got out, CSPOT does connect but same, nothing out.

Any ideas to debug thoses issues ?

rhaamo, 2022-03-08:
Apparently the 2.2 3378 has I2C on different pins... https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/26#issuecomment-967341319
- SDA: 18
- SCL: 23
That does makes the write failed goes away, but still no audio output, wondering if I2S is routed differently, or the headphones are on a different setting than "amp" out ?

rhaamo, 2022-03-08:
Got the artifact from the latest master CI build (feat MA12070p driver, after that the builds fails)

Selected the board "AI-Thinker Audio board (A1S ES8388)" again and now it works ? idk what have changed between the two builds but well...

Remain things are:
- heaphones jack has no output
- is line-in passthrough supported or planned ?

feelfreelinux, 2022-03-08:
Hi, thanks for trying this out! The A1S chip is honestly a mess - I was able to count 3 different versions of those chips with different DAC (AC101 or ES8388) and even different I2S pinouts. I have looked into the chip you most likely are using, and seems like it requires two extra registers (0x30 and 0x31) to have the volume written to in order to make it work. I have pushed the changes on newest master, will try to fix the build issue.

https://github.com/feelfreelinux/euphonium/commit/e26ee70af16602ae4bed48d94aca14974b988a19

Line-in support will come soon (currently planning t release along with multiroom and UI redesign). It will route all the audio from the aux in port, apply internal DSP and output on DAC.

Thanks for trying out euphonium btw! Feel free to reach out to me on `@feelfreelinux:matrix.org` with questions and feedback!

feelfreelinux, 2022-03-08:
Build fail resolved, corrupt cache issue.

The line-in support will come around the time when im able to assemble my board with MA12070P and WM8782 ADC. All part should be easily sourceable :)

rhaamo, 2022-03-08:
Ahah yes from what I have gathered in various links it's a mess, and it's even from a genuine board...

I have fetch the latest artifact and I can confirm that the headphones jack works ! But audio volume has no effect.

To plan if possible would be to be able to toggle headphones or amp output (like using both or only one of the two)

Thanks for planning line-in!

HeikoGr, 2022-03-10:
The toggle could be done via GPIO_NUM_12 as far as i know (https://github.com/pschatzmann/arduino-audiokit has got it working). it's on my todo, if @feelfreelinux is not faster then me...

HeikoGr, 2022-03-10:
i have an ai thinker board v2.2 3478 with ESP32-A1S v2974
Headphone Jack: working (with Volume control)
LineOut: working (with Volume control)

jwktje, 2022-05-21:
I have a V2.2 with v2974 but setting volume on the headphone seems to have no effect.
It works fine but the volume is very low. How did you manage to get it to work @HeikoGr  ?

HeikoGr, 2022-05-23:
sorry for late reply.
it works for me (same board) - but i compiled euphonium from the latest sources for myself.
did you use a release version?

nvschilleman, 2022-07-16:
> i have an ai thinker board v2.2 3478 with ESP32-A1S v2974 Headphone Jack: working (with Volume control) LineOut: working (with Volume control)

Do you mind sharing your binaries? I also have a v2974 board but I do not manage to get this to work? Also, did you change DIP settings on the board?