muvox-io/euphonium #28: Getting Error MSG

catarrer, 2022-04-19:
Hi, 
I tried to install Euphonium on ESP32. 

getting this error after flashing from serial monitor: 

ELF file SHA256: 7733f97eac86787c

Re-entered core dump! Exception happened during core dump!
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5624
load:0x40078000,len:14344
load:0x40080400,len:3680
entry 0x40080674
[0;32mI (27) boot: ESP-IDF v4.4-367-gc29343eb94 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 15:16:38[0m
[0;32mI (27) boot: chip revision: 1[0m
[0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (38) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (47) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (91) boot:  3 storage          Unknown data     01 82 00010000 0007d000[0m
[0;32mI (98) boot:  4 factory          factory app      00 00 00090000 000aae60[0m
[0;32mI (106) boot:  5 app              OTA app          00 10 00140000 00298aa0[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (117) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (125) esp_image: segment 0: paddr=00140020 vaddr=3f400020 size=68024h (426020) map[0m
[0;32mI (286) esp_image: segment 1: paddr=001a804c vaddr=3ffbdb60 size=07120h ( 28960) load[0m
[0;32mI (297) esp_image: segment 2: paddr=001af174 vaddr=40080000 size=00ea4h (  3748) load[0m
[0;32mI (299) esp_image: segment 3: paddr=001b0020 vaddr=400d0020 size=1756cch (1529548) map[0m
[0;32mI (851) esp_image: segment 4: paddr=003256f4 vaddr=40080ea4 size=1cf48h (118600) load[0m
[0;32mI (900) esp_image: segment 5: paddr=00342644 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (915) boot: Loaded app from partition at offset 0x140000[0m
[0;32mI (915) boot: Disabling RNG early entropy source...[0m
[0;32mI (927) psram: This chip is ESP32-D0WD[0m
[0;31mE (927) psram: PSRAM ID read error: 0xffffffff[0m
[0;31mE (928) cpu_start: Failed to init external RAM, needed for external .bss segment[0m

abort() was called at PC 0x40081951 on core 0

Backtrace:0x40082489:0x3ffe3ba00x400968e9:0x3ffe3bc0 0x4009c365:0x3ffe3be0 0x40081951:0x3ffe3c50 0x40079418:0x3ffe3c80  |<-CORRUPTED

feelfreelinux, 2022-04-20:
What esp chip are you running? You need one with psram to have enough memory to run the system.

catarrer, 2022-04-20:
I am using the Adafruit Esp32 D1 mini with the ESP32-WROVER-32 module.

feelfreelinux, 2022-05-09:
Are you running the prebuilt version? Looks like some PSRAM configuration error

J-Wrobel, 2023-02-27:
I stumbled into something similar, used prebuilt binary from 0.3.0 release. Log and esptools chip info in attachment.[](url)
[log.txt](https://github.com/muvox-io/euphonium/files/10844756/log.txt)
[chip_info.txt](https://github.com/muvox-io/euphonium/files/10844758/chip_info.txt)