muvox-io/euphonium #41: Bug: Undefined symbols for architecture x86_64

TooDissing, 2022-06-02:
When trying to build the CLI target on an Intel based macbook I am getting the following linker error:

I tried with the build in `make`and the latest version of `gmake`(GNU Make 4.3). Both with the same result.

```
Undefined symbols for architecture x86_64:
  "_DH_get0_pub_key", referenced from:
      CryptoOpenSSL::dhInit() in libbell.a(CryptoOpenSSL.cpp.o)
  "_DH_set0_pqg", referenced from:
      CryptoOpenSSL::dhInit() in libbell.a(CryptoOpenSSL.cpp.o)
  "_EVP_CIPHER_CTX_reset", referenced from:
      CryptoOpenSSL::aesECBdecrypt(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >&) in libbell.a(CryptoOpenSSL.cpp.o)
  "_HMAC_CTX_free", referenced from:
      CryptoOpenSSL::sha1HMAC(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&) in libbell.a(CryptoOpenSSL.cpp.o)
  "_HMAC_CTX_new", referenced from:
      CryptoOpenSSL::sha1HMAC(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&) in libbell.a(CryptoOpenSSL.cpp.o)
  "_OPENSSL_init_crypto", referenced from:
      bell::TLSSocket::TLSSocket() in libbell.a(TLSSocket.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [CMakeFiles/euphoniumcli.dir/build.make:137: euphoniumcli] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:175: CMakeFiles/euphoniumcli.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

I am sure that I am just missing a step, but tried to follow the steps from the getting started documentation: https://feelfreelinux.github.io/euphonium/technical/dev-environment/#folder-extensions

feelfreelinux, 2022-06-02:
What OpenSSL version are you using? You need openssl version 1.1 linked into the path.

TooDissing, 2022-06-02:
> What OpenSSL version are you using? You need openssl version 1.1 linked into the path.

That might explain it. The version I use are 3.something. Will try to get 1.1 installed as well and then try again. Thanks!