muvox-io/euphonium #49: [BUG] Full crash when trying to play webradio

TooDissing, 2022-08-14:
When trying to start playback on the "Jazz" radio station a full crash occurs  - Searching webradio for "jazz" and selecting the one name "Jazz in the top left corner".

```
I [http] HTTPModule.cpp:75: Posting the event
D [core] Core.cpp:116: Got event
I [http] HTTPServer.cpp:330: Publishing event
I [webradio] WebRadioPlugin.cpp:49: Starting WebRadio
I [core] Core.cpp:50: Shutting down except webradio
I [http] HTTPServer.cpp:330: Publishing event
I [webradio] WebRadioPlugin.cpp:53: Got access to audio buffer
I [http] HTTPModule.cpp:59: Responding to connection
I (98992) I2S: APLL Enabled, coefficient: sdm0=28, sdm1=8, sdm2=5, odir=2
I [http] HTTPModule.cpp:60: Status: 200
I [http] HTTPModule.cpp:61: Body: {"status":"playing"}
D [core] Core.cpp:116: Got event
I [http] HTTPServer.cpp:330: Publishing event
Socket opened
I [webradio] HTTPAudioStream.cpp:74: Big error
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401a5260  PS      : 0x00060330  A0      : 0x801aaca5  A1      : 0x3ffd4950  
0x401a5260: tcp_input at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/core/tcp_in.c:251

A2      : 0x3f8bce64  A3      : 0x00000000  A4      : 0x00000002  A5      : 0x3f8c0556  
A6      : 0x8c0c0300  A7      : 0x00000011  A8      : 0x00000000  A9      : 0x3f8849d8  
A10     : 0x00000001  A11     : 0x3f8849d8  A12     : 0x00000070  A13     : 0x3f8a5688  
A14     : 0x2801a8c0  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8c0c033c  LBEG    : 0x40092be4  LEND    : 0x40092c00  LCOUNT  : 0xffffffff  
0x40092be4: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x40092c00: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace:0x401a525d:0x3ffd49500x401aaca2:0x3ffd49c0 0x401aed7a:0x3ffd49f0 0x4019fb75:0x3ffd4a10 
0x401a525d: tcp_input at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/core/tcp_in.c:238

0x401aaca2: ip4_input at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/core/ipv4/ip4.c:784

0x401aed7a: ethernet_input at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/netif/ethernet.c:186

0x4019fb75: tcpip_thread_handle_msg at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at /home/tue/esp/esp-idf_4.4-rc1/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 1b6f4c08907580c2

Initiating core dump!
I (8562) esp_core_dump_uart: Press Enter to print core dump to UART...
I (8569) esp_core_dump_uart: Print core dump to uart...
Core dump started (further output muted)
Received  45 kB...
Core dump finished!
Failed to run espcoredump script: Command '['/home/tue/.espressif/python_env/idf4.4_py3.8_env/bin/python', '/home/tue/esp/esp-idf_4.4-rc1/tools/idf_monitor_base/../../components/espcoredump/espcoredump.py', 'info_corefile', '--core', '/tmp/tmpz74cr5ep', '--core-format', 'b64', '/home/tue/code/opennext/stroem/software/euphonium/targets/esp32/app/build/euphonium-esp32.elf']' returned non-zero exit status 1.
b'espcoredump.py v0.4-dev\nTraceback (most recent call last):\n  File "/home/tue/esp/esp-idf_4.4-rc1/tools/idf_monitor_base/../../components/espcoredump/espcoredump.py", line 350, in <module>\n    temp_core_files = info_corefile()\n  File "/home/tue/esp/esp-idf_4.4-rc1/tools/idf_monitor_base/../../components/espcoredump/espcoredump.py", line 149, in info_corefile\n    core_elf_path, target, temp_files = get_core_dump_elf(e_machine=exe_elf.e_machine)\n  File "/home/tue/esp/esp-idf_4.4-rc1/tools/idf_monitor_base/../../components/espcoredump/espcoredump.py", line 68, in get_core_dump_elf\n    loader = ESPCoreDumpFileLoader(args.core, args.core_format == \'b64\')\n  File "/home/tue/esp/esp-idf_4.4-rc1/components/espcoredump/corefile/loader.py", line 564, in __init__\n    self._get_core_src(path)\n  File "/home/tue/esp/esp-idf_4.4-rc1/components/espcoredump/corefile/loader.py", line 582, in _get_core_src\n    data = base64.standard_b64decode(line.rstrip(b\'\\r\\n\'))\n  File "/usr/lib/python3.8/base64.py", line 105, in standard_b64decode\n    return b64decode(s)\n  File "/usr/lib/python3.8/base64.py", line 87, in b64decode\n    return binascii.a2b_base64(s)\nbinascii.Error: Invalid base64-encoded string: number of data characters (25) cannot be 1 more than a multiple of 4\n'
```

[coredump.20220814.txt](https://github.com/feelfreelinux/euphonium/files/9333011/coredump.20220814.txt)

TooDissing, 2022-08-14:
Trying to reproduce, but not so easy it seems.

When starting plyback of either webradio or spotify, the playback starts, but quickly crashes after a few seconds and tuning into what sounds like pink noise. Not ideal.

Any pointers?

feelfreelinux, 2022-08-19:
Hi,

This looks really similar to the memory corruption issue that currently happens with the LED driver. Are you using the volume knob peripherals?

TooDissing, 2022-08-20:
> Hi,
> 
> This looks really similar to the memory corruption issue that currently happens with the LED driver. Are you using the volume knob peripherals?

No, not using anything else than the main board. Was able to flash an earlier version of the code and then get it to work. So likely something within the code or some of the .be scripts.

However I have no clue on where to look next, as the same commit did work on an earlier flash.

Maybe something happens within the build? 
How to ensure all generated resources are cleared completely?