muvox-io/euphonium #55: the .vscode folder should be excluded from this repo

jdthorpe, 2022-09-20:
The `.vscode` folder at the root of this repo has a setup that is particular to the original author's computer.  For others, this won't have the correct setup and will be missing things like the proper includes paths for resources provided by the ESP-IDF framework (e.g. `freertos`, `driver/gpio`, etc).  To be clear, this won't stop the code from compiling, but the IDE won't know about any of the resources provided by the ESP-IDF (e.g. header files), resulting in lots of errors in the IDE like `cannot open source file "freertos/queue.h"`.

VS code users who wish to set up the proper includes directories for the ESP-IDF framework should:
- install the ESP-IDF VS Code extension
- install the ESP IDF through the extension or link the extension to an existing ESP-IDF installation
- delete the `.vscode` folder included in this repo
- open the command pallet in VS CODE and select "ESP-IDF: Add VS Code Configuration Folder" to create the dependencies that are appropriate for the ESP-IDF installation on your machine

See the [ESP-IDF Extension docs](https://github.com/espressif/vscode-esp-idf-extension) for details.

HeikoGr, 2022-09-20:
You can blame me for this.

i thought it would be convenient to have at least the recommendations from .vscode/extensions.json

jdthorpe, 2022-09-20:
No worries.  I mostly posted this b/c I couldn't find a good reference on how to get VSCode to play nice with an ESP-IDF project not set up via their plugin, and thought this was as good a place to mention it -- especially since it was this project I was playing with :) 

The "ESP-IDF: Add VS Code Configuration Folder" command won't update an existing `.vscode/c_cpp_properties.json` which is where all the includes paths are added.  Perhaps just a mention of the steps above in the [docs/techincal/dev-environment.md](https://github.com/feelfreelinux/euphonium/blob/master/docs/technical/dev-environment.md) for VS Code users would be the right bit of nudging.