muvox-io/euphonium #71: CSpot issues + questions about building

kuba2k2, 2023-04-01:
Hi

I've been testing Euphonium (`master` branch, 3e70f598d55f137aaeaf42e023980990d8230eaa) on ESP32 for a bit. I built it on Ubuntu to have the somewhat-latest version (but IIRC the experience was the same in the released 0.3.0 build). There are some issues with CSpot that make the thing pretty hard to use:

- there are issues connecting using Zeroconf - my smartphone doesn't see it at all, my other smartphone sees it but can't really connect, and the only way that it works is from my PC (after connecting I can control it from my smartphone)
- volume control doesn't work - it's at 100% the whole time (#70)
- queue doesn't work - anything added to the queue is simply ignored and skipped to the next track
- moving queue items is not possible - I can't seem to be able to even "drag" a song to change its position
- clicking a track in the queue (i.e. skipping 3 tracks forward) doesn't keep the previous playback context - so when I'm listening to a playlist in shuffle mode, skipping to a track changes all the following tracks, which generate randomly (including the ones that have already played before)
- there's a weird "scratching" sound between tracks (the beginning and end of tracks are breaking)
- track crossfading doesn't work (on albums that support it), there's a 1-or-so second pause instead
- seeking backwards breaks the playback (only tried it once, but the device stopped playing and disappeared from the Spotify Connect list)

I know some of these issues are very minor, but I'm wondering if they have been (or are planned to be) fixed, on some other branch.

About that... I wasn't able to build the `develop` branch, since it's Nix based, and I have no idea how to run that, nor haven't found any docs that would guide me to do it. Any help would be appreciated :smile: 

I also don't fully understand the choice of using Nix, since IMHO it's quite complicated (especially for new users) and different from all build systems I've seen so far - after trying various commands (including some opt-in "experimental" ones), it did even download ESP-IDF but the build never actually happened. Apart from that, it sadly doesn't support Windows /without WSL/, which makes it pretty much impossible for a developer like me to work with it.

I don't mean to be trying to convince you guys to other build systems, but I honestly think that something like PlatformIO (without Arduino, of course) would be a much better fit - it's especially meant for embedded systems (but can target host-native as well), it's very easy to integrate into a project/library, and it's simple to understand and build with it.

philippe44, 2023-04-01:
If I may, as a dev who is using cspot, the core of euphonium, arduino is a very simplified system which is not at all capable of building complex things like euphonium. It's more a tinkering platform and I would personally never even try. Now, to your question, I've made a set of PR from which most of them have been accepted by @feelfreelinux to make cspot compile and run under Windows, so if you install the espressif environment for Win and VisualStudio, it is likely to work for the whole euphonium (although I'm using it myself only for cspot)

kuba2k2, 2023-04-02:
I think you misunderstood me a little bit. I know that esp-idf doesn't need WSL - I was perfectly able to compile CSpot and more in CLion using Cygwin - before your Windows support PR, after that it sadly stopped working. I know I could *probably* install Visual Studio, but I dislike the IDE very much and don't have enough space for all the C++ SDKs and workloads.

I was saying that Nix needs WSL on Windows - last time I checked, there was no native binaries for Win32. PlatformIO itself has pretty much nothing to do with Arduino, if you don't include the Arduino frameworks. It's simply a package manager and build system, which handles installation of esp-idf, toolchains, libraries, etc. I can agree that Arduino IDE is a tinkering platform, but those are totally different things.

After all, both Euphonium and CSpot (and bell) are just collections of header and source files, which is not such a complex thing to build. I guess that if you just grabbed all .cpp and .h files, put them in one directory, you'd be able to compile it all with little modifications.

Anyway, I'm mostly asking about the runtime issues and some guidance on how to compile the develop branch - I don't need any other build systems at present. When the time comes to embed CSpot in my project, I'll probably port it to platformio anyway.

philippe44, 2023-04-02:
Indeed I misunderstood you. EhY did the Windows PR break? Is it still the case with the latest PR in the dev/cdn branch?

kuba2k2, 2023-04-02:
I don't remeber exactly, but it was something related to OpenSSL I think. Also, you've added PortAudio as a precompiled library, which is incompatible with Cygwin and works with Visual Studio only.

Normally, in Cygwin, I was installing OpenSSL and PortAudio using Cygwin's package manager, and CMake was able to find them and compile everything; after the PR that stopped working.

philippe44, 2023-04-02:
OpenSSL is probably a different item as I did not touch it, @feelfreelinux moved to only mbedtls if I remember. Now, I added support for it in Win32. For portaudio, there is a Findportaudio.cmake that  can be changed. Do you still need that port and shall I have a look at this or this is not the case anymore? Ultimately, what platform do you plan using?

philippe44, 2023-04-02:
What I also mean is that in some of the cmake or related files, I might have used win32 and msvc flags interchangeably which is not correct obviously

kuba2k2, 2023-04-02:
I'm not currently planning to be developing using bell or Euphonium, but when the time comes I'll still be developing on Windows, in CLion or VSCode, targeting ESP32 and testing on Win32. Again, you don't have to worry about these issues as I'll port this to PIO for myself. I only wanted to build the `develop` branch to have fixes for some usability issues, but it's not CMake based and I don't know how to build it.

philippe44, 2023-04-02:
ok then - btw, when you mean it's not cmake based, which part are you referring to exactly?

kuba2k2, 2023-04-02:
The `develop` branch of Euphonium is using Nix for building instead of CMake (see #59).

philippe44, 2023-04-02:
oh, I missed that entirely as I only use cspot/bell and rely on raw cmake