muvox-io/euphonium #79: Latest release is 2 years old

anabolyc, 2024-05-10:
Hey folks, I was curios to install the latest version of the project to try it out on the [Esparagus boards](https://www.crowdsupply.com/sonocotta/esparagus-media-center) that I've designed. I just noticed that latest release 0.3.0 is 2 years old and have troubles using it (cannot pass wifi configuration). I was thinking that maybe latest release would solve those issues, but build system is quite complex to just quickly "give it a try".

I guess I'm asking if it is just a matter of tagging the latest stable build with specific version, or is it really the latest stable code that I can try out.

alufers, 2024-05-11:
Hi, sorry. We are currently still developing euphonium and it is not ready yet for public use. There probably will be problems running it on a different DAC/amplifier. than the one built into the muVox. Please stand by. The project is not abandoned, but we are taking our time.

agdamsbo, 2024-05-31:
@anabolyc I was suggested to try this fork and build for your boards. Haven't had the chance yet, but here you go: https://github.com/LeoSum8/euphonium_louderESP/releases/tag/develop_240510_louder